pimoroni/keybow-firmware #1: Debugging Keyboard Layouts via Wifi?

TheDiveO, 2018-12-18:
Since the keybow hw uses a Raspi 0WH, would it be possible to edit and test keyboard layouts via WiFi while the keyboard is acting as HID via USB? Is it possible to smb mount the keyboard layout lua scripts on the host via WiFi?

dglaude, 2018-12-19:
This is a bit dangerous (to have a wifi hackable keyboard).
Actually, using a "Raspberry Pi Zero WH" not because of the W but because of the H is a dangerous choice.

Keybow is based on Tiny Core Linux (piCore), but rather than to use two partition, on FAT and one linux to put additional software and persistance, it put everything in into the FAT partition, so accessible from /boot/.

So if you want to add wifi connectivity you could try following tutorial dedicated to the Pi Zero W such as:  https://www.novaspirit.com/2018/01/09/tiny-core-raspberry-pi-zero-w-install/
There are a few video that explain the process such as: https://www.youtube.com/watch?v=aKvW59uk4PY

In particular, you will need at least two modules: firmware-rpi3-wireless and wifi

Once you have successfully done that on piCore, then you could try to apply that knowledge and process to keybow. The most obvious result I could see is that you would be able to SSH into your PiZero, from there edit the configuration file and reboot.

There are other way to access the microSD content, without Wifi, such as enabling gadget_ethernet on the same USB and ssh from the host where keybow is connected. The last way is to have gadget_storage and be able to access the microSD as a USB key... the issue beeing to switch from Storage mode to Keyboard mode (or you could have both simultaneously.

One of the problem is that the source code to build the SD card is not provided, nor is there building instruction/script/. So you may have to end-up reverse engineer what is just compiled/installed free software...

Gadgetoid, 2018-12-21:
Actually it's a custom build of Raspbian, rather than piCore (the piCore details were a miscommunication between me and Sandy). But there's no reason why it couldn't run either on:

1. Full-blown Raspbian. The statically-linked keybow binary will "just work".
2. piCore - again, I suspect the keybow binary will just work

I actually spent a lot of time debugging Keybow over WiFi during the early development stages, but I later ripped out almost everything in an effort to produce the smallest "initrd" possible to speed up boot times.

As @dglaude suggests, there's definitely something to be said for the security of having a WiFi-enabled keyboard device though :D Albeit I do half of my work directly *on* Pi's so I'm not sure I'd be any worse off.

TheDiveO, 2018-12-29:
I don't really see the point against Wifi for debugging during keyboard setup. I agree that for normal use, minimizing the attack surface is always not bad. And then considering the disaster of the cheap middle kingdom wireless keyboards with totally broken "encryption", then even Wifi WPA2 seems yet to be worlds apart.

TheDiveO, 2019-01-07:
Since I managed to come up with a Lua unit test setup to develop my Keybow keymaps on a Ubuntu-based host, in-target debugging isn't necessary for me anymore. I'm thus closing this issue.

For those interested how my test setup looks like, please check out [thediveo/multibow](https://github.com/TheDiveO/multibow).

closed 2019-01-07 by TheDiveO