pimoroni/keybow-firmware #3: Strongly QWERTY minded without notice nor option to change that...

dglaude, 2018-12-19:
Because keyboard interface is not specifying the keyboard mapping, Keybow/LUA software seems to assume you use QWERTY in UK or US (to be verified) setting.

With my MAC configured for AZERTY Belgian FR, I get "strange" result.

So if I say that a key should send an "m" it will send a ","

If I use the following to produce a "qwaz" by pressing key 00, I get a "azqw":
function handle_key_00(pressed) -- Lorem Ipsum
    if pressed then
        keybow.text([[
qwaz
]])

So the "Lorem ipsum dolor..." does not look the same anymore.

What is needed is some kind of "reverse mapping" (a set of reverse mapping to choose from) that can be applied to everything you request. The default mapping would be "no mapping" and work for UK or US keyboard. That way, any working script can be just modified for your specific mapping.

PS: I don't have a keybow hardware, I just tested the software with a ZeroSeg that has two button, one that is "key_00".

dglaude, 2018-12-19:
Maybe this piece of mapping can be reused to customize for specific kind of keyboard: https://github.com/whid-injector/WHID/tree/master/tools/keyboard_layouts

This is C code, so it could be easy to integrate into keybow source code.

Gadgetoid, 2018-12-21:
The key layout should be possible to change entirely in LUA, without messing with C. I'd had a suspicion that alternate region key layouts would cause us some hassle but I didn't have a good way to deal with them.

Those layout files you link look interesting- however- and make a strong case for supplying a C-function for translating ASCII to HID codes + modifier keys. I wonder if they could be sensibly ported to LUA- my current approach looks something like this:

```lua
local KEYCODES         = "abcdefghijklmnopqrstuvwxyz1234567890\n\a\b\t -=[]\\#;'`,./"
local SHIFTED_KEYCODES = "ABCDEFGHIJKLMNOPQRSTUVWXYZ!@#$%^&*()\a\a\a\a\a_+{}|~:\"~<>?"
```
And replacing these two lines with a region-specific alternative should go a long way toward making it function correctly for a given region.

Gadgetoid, 2019-01-08:
Yes, it's a basic "boot" style keyboard that sends standard HID keycodes.

It looks like- from your quoted description- that Keybow isn't doing anything wrong per se, but I suspect any given OS only allows *one* Keyboard type to be specified, which is then applied across all connected keyboards?

That's kind of unfortunate, since it's not inherently portable from one system (with one language) to another, but it does make a case for including some sort of USB Serial control over Keybow (since you might want to change Language on the fly).

I think we need to change how the mapping table is expressed in Keybow.lua and move all mapping out into an language/layout-coded include file. I don't have much visability on this problem since I use only QWERTY/English keyboards (ho boy my geek cred just docked 100 points), so any and all feedback on how to make this as painless as possible is appreciated.

TheDiveO, 2019-01-08:
I don't think (and never thought) that the Keybow isn't doing anything correctly here -- but instead this is an architectural design of the USB HID keyboard spec.

As I'm using German keyboards I would be interested in a solution; it's not a burning issue atm, as most keys the Keybow needs to press for me are function keys and letters that are in the same place, but then even Z<->Y will need a localized keymap.

cvGimse, 2020-09-21:
Hi @Gadgetoid,
I'm having issues getting my keybow to work with my Norwegian keyboard. I really need it work as I'm using it to input code snippets and regex replaces. I've tried changing the local KEYCODES but without knowing what to replace I can't get it to work. I'm getting really weird results now. Need to type " ' * \. I can do a couple of them with workarounds, but have not yet been able to type ' and \. Is there any documentation for which keys the symbols in KEYCODES represent? And what do I do about keys that exist on US/UK querty but not on NO qwerty?

Gadgetoid, 2020-09-22:
It's a tricky problem (tedius, but not super difficult)- the `KEYCODES` and `SHIFTED_KEYCODES` values correspond to HID "usage" codes.

The resource I used to figure out what these tables should be is: https://gist.github.com/MightyPork/6da26e382a7ad91b5496ee55fdc73db2

Keybow gets the index of a character in this lookup table (for example `b` is 2 (lua is 1-indexed!)) and adds 3 to get the HID usage code. In this case the usage code for `b` is `2 + 3` or `5` which you can see here: https://gist.github.com/MightyPork/6da26e382a7ad91b5496ee55fdc73db2#file-usb_hid_keys-h-L41

Now as far as I'm aware- and here's where me being British English makes me uniquely hopeless at tackling this particular problem- a Norwegian or German keyboard produces exactly the same scancode and HID usage code for any key in a given position.

In this case the Norwegian (http://kbdlayout.info/KBDNO) and the United Kingdom (http://kbdlayout.info/kbduk) keyboards both have "a" in the same place, so it's fine.

Now if I fast-forward to `'` and `*` (I love how those look like tiny keyboard keys, ha) we find that `*` is in the key nestled in with Enter on the NO board which corresponds to `~` on the UK layout.

So we can change one of the ~ chars in `SHIFTED_KEYCODES` (and now I must admit I don't know which one so I'll do both):

```
local SHIFTED_KEYCODES = "ABCDEFGHIJKLMNOPQRSTUVWXYZ!@#$%^&()=\a\a\a\a\a_+{}|*:\"*<>?"
```

I've also moved `()` one place to the left, and inserted `=` since I think that's the right place for those.

Note: the `\a` in the `SHIFTED_KEYCODES` counts as a single character.

The basic process is to repeat these steps, moving each character into the position of its corresponding keyboard usage code from https://gist.github.com/MightyPork/6da26e382a7ad91b5496ee55fdc73db2, in either the `SHIFTED` or non shifted list.

Hopefully with your knowledge of Norwegian keyboards and my knowledge of fumbling through things with blind determination we'll get something working.

cvGimse, 2020-09-23:
Hi,
Thank you for your reply! This might be a stupid question, but are you using the \a to denote dead keys, or what is its function? If so I've made a go at reproducing a Norwegian keyboard that looks OK, but I won't be able to test it until tomorrow.

```
local KEYCODES         = "abcdefghijklmnopqrstuvwxyz1234567890\n\a\b\t +\\å\a<\'øæ|,.-"
local SHIFTED_KEYCODES = "ABCDEFGHIJKLMNOPQRSTUVWXYZ!\"#¤%&/()=\a\a\a\a\a?\aÅ\a>*ØÆ§;:_"
```

That leads me to another question. As the Norwegian keyboard has three extra letters on it (æøå), it loses some space which means that brackets are moved to AltGr positions, making it a very frequently used key for programmers. { = AltGr + 7, [ = AltGr + 8, ] = AltGr +9, } = AltGr + 0. Will there be any support for AltGr in Keybow?

Could there be (for Norwegian in this case) a line saying:

```
local ALTGRED_KEYCODES = \a\a\a\a\a\a\a\a\a\a\a\a\a\a\a\a\a\a\a\a\a\a\a\a\a\a\a@£$€\a{[]}\a\a\a\a\a\a\a\a\a\a\a\a\a\a\a\a\a
```

And then could one change the keybow.set_key function to add these somehow? The AltGr would also need its own function I suppose. And what is the difference between LEFT_ALT and RIGHT_ALT in Keybow anyway?

Sorry for dumping a whole bunch of questions suddenly! I really appreciate you taking time to answer this.

Chris

Gadgetoid, 2020-09-23:
"\a" is, indeed, used for denoting unused chars. I needed *something* that was non-printable to fill that gap and the choice is pretty arbitrary- though I cannot recall my frame of mind when making it!

You are right about `ALTGRED_KEYCODES`, in order to reproduce those characters Keybow would have to "press" AltGr the same way it does the shift key currently.

In keybow `Left Alt` is just Alt, whereas `Right Alt` is- I think- actually `AltGr`. I guess the "GR" stands for "Goes on the Right" or something.

The magic function that decodes characters into keycodes could be changed to something like this:

```lua
function keybow.set_key(key, pressed)
    if type(key) == "string" then
        local shifted = SHIFTED_KEYCODES:find(key, 1, true)
        local altgrd = ALTGRED_KEYCODES:find(key, 1, true)
        local normal = KEYCODES:find(key, 1, true)

        local hid_code = shifted or altgrd or normal

        if not (hid_code == nil) then
            hid_code = hid_code + 3
            if shifted then keybow.set_modifier(keybow.LEFT_SHIFT, pressed) end
            if altgrd then keybow.set_modifier(keybow.RIGHT_ALT, pressed) end
            keybow_set_key(hid_code, pressed)
        end

    else -- already a key code
        keybow_set_key(key, pressed)
    end
end
```

To test this outside of Keybow (which is exhaustingly difficult) you can use repl.it's Lua interpreter - https://repl.it/languages/lua

Here's the mocked (very broken to not be keybow compatible directly) code I used in repl.it to test that the above could work:

```lua
local KEYCODES         = "abcdefghijklmnopqrstuvwxyz1234567890\n\a\b\t -=[]\\#;'`,./"
local SHIFTED_KEYCODES = "ABCDEFGHIJKLMNOPQRSTUVWXYZ!@#$%^&*()\a\a\a\a\a_+{}|~:\"~<>?"
local ALTGRED_KEYCODES = "\a\a\a\a\a\a\a\a\a\a\a\a\a\a\a\a\a\a\a\a\a\a\a\a\a\a\a@£$€\a{[]}\a\a\a\a\a\a\a\a\a\a\a\a\a\a\a\a\a"

LEFT_CTRL = 0
LEFT_SHIFT = 1
LEFT_ALT = 2
LEFT_META = 3
RIGHT_CTRL = 4
RIGHT_SHIFT = 5
RIGHT_ALT = 6
RIGHT_META = 7

function set_key(key, pressed)
    print("------------")
    print("Setting Key", key, pressed)
    if type(key) == "string" then
        local shifted = SHIFTED_KEYCODES:find(key, 1, true)
        local altgrd = ALTGRED_KEYCODES:find(key, 1, true)
        local normal = KEYCODES:find(key, 1, true)

        local hid_code = shifted or altgrd or normal

        if not (hid_code == nil) then
            hid_code = hid_code + 3
            if shifted then keybow_set_modifier(LEFT_SHIFT, pressed) end
            if altgrd then keybow_set_modifier(RIGHT_ALT, pressed) end
            keybow_set_key(hid_code, pressed)
        else
            print("Character not found!")
        end

    else -- already a key code
        keybow_set_key(key, pressed)
    end
    print("------------")
end

function keybow_set_modifier(modifier, pressed)
  if modifier == 1 then
  print("Shift",pressed)
  end
  if modifier == 6 then
  print("AltGr",pressed)
  end
end

function keybow_set_key(key, pressed)
    print("KeyCode", key, pressed)
end

set_key("a", true)
set_key("A", true)
set_key("é", true)

set_key("@", true)
set_key("£", true)
set_key("$", true)
set_key("€", true)
```

This same pattern should work for any number of function keys but doesn't currently support something like `É` which requires `Shift + AltGr + e` on my keyboard. I don't know if Norwegian has any cases like that?

I know this post is already overwhelmingly full of huge code dumps, but I managed to get Shift + AltGr combos working like so:

```lua
local KEYCODES         = "abcdefghijklmnopqrstuvwxyz1234567890\n\a\b\t -=[]\\#;'`,./"
local SHIFTED_KEYCODES = "ABCDEFGHIJKLMNOPQRSTUVWXYZ!@#$%^&*()\a\a\a\a\a_+{}|~:\"~<>?"
local ALTGRED_KEYCODES = "á\a\a\aé\a\a\aí\a\a\a\a\aó\a\a\a\a\aú\a\a\a\a\a\a@£$€\a{[]}\a\a\a\a\a\a\a\a\a\a\a\a\a\a\a\a\a"

local SHIFALTGRED_KEYCODES = "Á\a\a\aÉ\a\a\aÍ\a\a\a\a\aÓ\a\a\a\a\aÚ\a\a\a\a\a\a@£$€\a{[]}\a\a\a\a\a\a\a\a\a\a\a\a\a\a\a\a\a"

LEFT_CTRL = 0
LEFT_SHIFT = 1
LEFT_ALT = 2
LEFT_META = 3
RIGHT_CTRL = 4
RIGHT_SHIFT = 5
RIGHT_ALT = 6
RIGHT_META = 7

function set_key(key, pressed)
    print("------------")
    print("Setting Key", key, pressed)
    if type(key) == "string" then
        local shifted = SHIFTED_KEYCODES:find(key, 1, true)
        local altgrd = ALTGRED_KEYCODES:find(key, 1, true)
        local shiftaltgrd = SHIFALTGRED_KEYCODES:find(key, 1, true)
        local normal = KEYCODES:find(key, 1, true)

        local hid_code = shiftaltgrd or shifted or altgrd or normal

        if shiftaltgrd then -- short circuit shifted + altgrd to true for shiftaltgrd
          shifted = true
          altgrd = true
        end

        if not (hid_code == nil) then
            hid_code = hid_code + 3
            if shifted then keybow_set_modifier(LEFT_SHIFT, pressed) end
            if altgrd then keybow_set_modifier(RIGHT_ALT, pressed) end
            keybow_set_key(hid_code, pressed)
        else
            print("Character not found!")
        end

    else -- already a key code
        keybow_set_key(key, pressed)
    end
    print("------------")
end

function keybow_set_modifier(modifier, pressed)
  if modifier == 1 then
  print("Shift",pressed)
  end
  if modifier == 6 then
  print("AltGr",pressed)
  end
end

function keybow_set_key(key, pressed)
    print("KeyCode", key, pressed)
end

set_key("a", true)
set_key("A", true)
set_key("é", true)
set_key("É", true)

set_key("@", true)
set_key("£", true)
set_key("$", true)
set_key("€", true)
```

I'll make these changes to `keybow.lua`.

cvGimse, 2020-09-23:
Hi again,
Wow, thank you! Mind that the AltGr sequence from the above code fits an NO keyboard, not UK/US. You would know this better than me, but I think UK/US would look like this:
```
local ALTGRED_KEYCODES = "á\a\a\aé\a\a\aí\a\a\a\a\aó\a\a\a\a\aú\a\a\a\a\a\a\a\a€\a\a\a\a\a\a\a\a\a\a\a\a\a\a\a\a\a\a\a\a\a\a\a"
local SHIFALTGRED_KEYCODES = "Á\a\a\aÉ\a\a\aÍ\a\a\a\a\aÓ\a\a\a\a\aÚ\a\a\a\a\a\a\a\a€\a\a\a\a\a\a\a\a\a\a\a\a\a\a\a\a\a\a\a\a\a\a\a"
```
Thanks again!

Chris

Gadgetoid, 2020-09-23:
You're right- I'm preparing a pull request to add these to Keybow. Just giving my code a quick test first!

I decided to swap the padding char, since it was hurting my eyes:

```
local KEYCODES             = "abcdefghijklmnopqrstuvwxyz1234567890\n•\b\t -=[]\\#;'`,./"
local SHIFTED_KEYCODES     = "ABCDEFGHIJKLMNOPQRSTUVWXYZ!@#$%^&*()•••••_+{}|~:\"~<>?"
local ALTGRD_KEYCODES      = "á•••é•••í•••••ó•••••ú••••••••••••••••••••••••••••••••"
local SHIFTALTGRD_KEYCODES = "Á•••É•••Í•••••Ó•••••Ú••••••••••••••••••••••••••••••••"
```

Right now, though, all I'm getting out of my attempt to type the text "áéíóú" and "ÁÉÍÓÚ" with `keybow.text()" is "ÁÁÁÁÁÁÁÁÁÁ". It's... a start... I guess.

Gadgetoid, 2020-09-23:
Okay using `•` was a catastrophically bad idea because Lua does funny things with Unicode in strings. `\a` it is!

cvGimse, 2020-09-23:
That was your frame of mind right there probably!

Gadgetoid, 2020-09-23:
Okay I can finally produce áéíóú and ÁÉÍÓÚ on my keybow without it melting. That's a good start.

I had to change some of the very deep and scary plumbing to support AltGr and AltGr+Shift modes. Lua treats Unicode strings as sequences of bytes so everything from finding a particular unicode character, to iterating through individual characters in a string becomes horribly, horribly broken.

I have switched to using Tables (Lua's idea of an array or list) to store the individual characters for lookup, since it clearly delineates between characters and makes it easier to convert a specific one (be it unicode or not) to an index and thus a keyboard code.

See: https://github.com/pimoroni/keybow-firmware/pull/61

drop0x55, 2020-09-26:
This works like a dream @Gadgetoid . I copied the changes to keybow.lua and substituted all keycodes with NO keycodes and it runs flawlessly. Nice work!

Gadgetoid, 2020-10-05:
Thanks for your help here @drop0x55 (not to mention doing the legwork of making the keyboard layouts) - I think we can finally close this nearly two year old issue!

closed 2020-10-05 by Gadgetoid