pimoroni/keybow-firmware #21: Non-US key code support for # ~ | and \ plus small shift-state bugfix

Gadgetoid, 2019-03-14:
See here for details and the "fixed" `set_key` method: https://forums.pimoroni.com/t/keybow-running-a-desktop-shortcut/10140

Need to find a better, user-configurable way of emitting the right keycodes for the right regions- the fix mentioned in the forums currently uses hard-coded edge cases to account for `# ~ | \`.

Moving the shift key release until after the key release is a minor bugfix to avoid weird casing issues when outputting text.

TheDiveO, 2019-04-03:
fyi, I'm working on international keyboard layout support in Multibow, supporting also altgr and hopefully dead key sequences, such as for Swiss sg/sf layouts. layouts are comparably easily defined as a layout, with the Multibow runtime calculating the glyph to keycode and modifier mapping itself from the layout data.

Gadgetoid, 2020-10-01:
#63 should finally put this to rest... I think! I'm not actually totally sure.