pimoroni/keybow-firmware #22: Degrees Symbol and +/- Symbol

gavinhill1973, 2019-04-18:
Hi,
I was wondering if you can help. I'm trying to use Keybow to add a degrees C and =/- symbols to documents using macros of Alt and keypad numbers.

Alt 248 and Alt 241. Code Copied in below. For the +/- I get an double entry one on key press and one on key release. The degrees C symbol seems to jump back to the start of the line.

Any guidance would be great.

Thanks

Gavin

require "keybow"

-- Standard number pad mapping --

-- Key mappings --
state = 0

function setup() -- Set custom lights up
keybow.auto_lights(false)
keybow.clear_lights()
state = 0
keybow.set_pixel(2, 255, 0, 0) -- Red
end

function handle_key_00 (pressed)
	 if state == 0 then
    keybow.set_key("0", pressed)
	 if pressed then
        keybow.set_pixel(0, 255, 255, 255)
    else
        keybow.set_pixel(0, 0, 0, 0)
	end
	end
	if state == 1 then
	        keybow.set_modifier(keybow.LEFT_ALT, keybow.KEY_DOWN)
            keybow.tap_key(keybow.KP2)
			keybow.tap_key(keybow.KP4)
			keybow.tap_key(keybow.KP8)
            keybow.set_modifier(keybow.LEFT_ALT, keybow.KEY_UP)
			keybow.set_key("C", pressed)
		end
end

function handle_key_01(pressed)
    if state == 0 then
   		 keybow.set_key(".", pressed)
		  if pressed then
        keybow.set_pixel(1, 255, 255, 255)
    else
        keybow.set_pixel(1, 0, 0, 0)
		end
	end
	if state == 1 then
	        keybow.set_modifier(keybow.LEFT_ALT, keybow.KEY_DOWN)
            keybow.tap_key(keybow.KP2)
			keybow.tap_key(keybow.KP4)
			keybow.tap_key(keybow.KP1)
            keybow.set_modifier(keybow.LEFT_ALT, keybow.KEY_UP)
				end
end

function handle_key_02(pressed)
   if pressed then
	state = state + 1
	if state == 1 then
	keybow.set_pixel(2, 0, 255, 0) -- Green
	end
	if state == 2 then
	keybow.set_pixel(2, 0, 0, 255) -- Blue
	end
	if state == 3 then
	keybow.set_pixel(2, 255, 0, 0) -- Red
	state = 0
	end
	keybow.sleep(500)
end
end

Gadgetoid, 2019-04-29:
Could you wrap your code paste in code tags? It's a bit hard to parse right now. Either using the <> in the editor, or by manually formatting in backticks like so:

    ```
    code goes here
    ```

gavinhill1973, 2019-04-29:
Hi
Thank you for getting back to me. 
I may be trying to run before I can walk but what I'm trying to do is set up some macros to enter the following °C  by pressing Left ALT and typing on the numeric keypad the numbers 2 4 8 then releasing the ALT key to get ° then I'm putting a C after it. Also to enter ±  by pressing Left ALT and typing on the numeric keypad the numbers 2 4 8 then releasing the ALT key.

I've gone back a few stages to try and get the basics to work but now I get °CC when I press key 0 and 
±±when I press key 1 one appears on key press and the other on release.

I've read through the https://learn.pimoroni.com/tutorial and I'm not sure where I'm going wrong.

I hope this makes more sense. Any help would be great as I think the key is a cool bit of hardware I just can not seem to get it to do this.

Thanks

Gavin

the default.lua code is pasted below

require "keybow"  
require "snippets/morekeys"

-- Standard number pad mapping --

-- Key mappings --

function handle_key_00(pressed)
		keybow.set_modifier(keybow.LEFT_ALT, keybow.KEY_DOWN)
		keybow.tap_key(keybow.KP2)
		keybow.tap_key(keybow.KP4)
		keybow.tap_key(keybow.KP8)
   		keybow.set_modifier(keybow.LEFT_ALT, keybow.KEY_UP)
		keybow.set_key("C", pressed)
		keybow.sleep(500)
end	

function handle_key_01(pressed)
  		keybow.set_modifier(keybow.LEFT_ALT, keybow.KEY_DOWN)
  		keybow.tap_key(keybow.KP2)
		keybow.tap_key(keybow.KP4)
		keybow.tap_key(keybow.KP1)
   		keybow.set_modifier(keybow.LEFT_ALT, keybow.KEY_UP)
		keybow.sleep(500)
end

function handle_key_02(pressed)
    keybow.set_key(keybow.ENTER, pressed)
end

function handle_key_03(pressed)
    keybow.set_key("1", pressed)
end

function handle_key_04(pressed)
    keybow.set_key("2", pressed)
end

function handle_key_05(pressed)
    keybow.set_key("3", pressed)
end

function handle_key_06(pressed)
    keybow.set_key("4", pressed)
end

function handle_key_07(pressed)
    keybow.set_key("5", pressed)
end

function handle_key_08(pressed)
    keybow.set_key("6", pressed)
end

function handle_key_09(pressed)
    keybow.set_key("7", pressed)
end

function handle_key_10(pressed)
    keybow.set_key("8", pressed)
end

function handle_key_11(pressed)
    keybow.set_key("9", pressed)
end

bertlea, 2019-05-10:
@gavinhill1973,  the "handle_key_xx" function will be triggered twice once when it is pressed and once is when it is released.  Thus, you may try to warp your key press sequence to run in the condition that only when pressed (not released).  e.g.:

function handle_key_00(pressed)
  if pressed then
    keybow.set_modifier(keybow.LEFT_ALT, keybow.KEY_DOWN)
    keybow.tap_key(keybow.KP2)
    keybow.tap_key(keybow.KP4)
    keybow.tap_key(keybow.KP8)
    keybow.set_modifier(keybow.LEFT_ALT, keybow.KEY_UP)
    keybow.set_key("C", pressed)
    keybow.sleep(500)
  end
end

function handle_key_01(pressed)
  if pressed then
    keybow.set_modifier(keybow.LEFT_ALT, keybow.KEY_DOWN)
    keybow.tap_key(keybow.KP2)
    keybow.tap_key(keybow.KP4)
    keybow.tap_key(keybow.KP1)
    keybow.set_modifier(keybow.LEFT_ALT, keybow.KEY_UP)
    keybow.sleep(500)
  end
end

gavinhill1973, 2019-05-16:
@bertlea,
Thank you very much for your help. That worked perfectly.

closed 2019-05-16 by gavinhill1973