pimoroni/keybow-firmware #23: Lack of documentation: Why 0b11100011?

mrusme, 2019-04-24:
As I simply could not find any documentation on this, I'm misusing GItHub issues to ask this question and also suggest documenting this:

https://github.com/pimoroni/keybow-firmware/blob/master/keybow/lights.c#L108

What's that magic number? Why not 255 instead of 227? 255 seems to do the job as well. And what's that field in first place? I first thought it's the brightness, however, while it does seem to make the LEDs brighter when running at 255 instead of 227, it does not seem to support all numbers.

Could you please give some insights on this?

Gadgetoid, 2019-04-24:
I'd suggest reading https://cpldcpu.wordpress.com/2014/08/27/apa102/ if you're interested in driving APA102 LEDs.

This magic number is the LED frame marker, which is indicated by 3 high bits set to 1: `0b11100000` plus 5 bits of global (applies to R, G and B LEds in a package) brightness: `0b00011111` (giving an 0-31 brightness range). Since we don't want to drive at maximum brightness, but still want colours to have the full 255*255*255 range, we use global brightness to handle the dimming across the LEDs.

From the above article:
[image]

mrusme, 2019-04-24:
@Gadgetoid thank you for clarification! This information is a great piece of documentation.