pimoroni/keybow-firmware #29: Occasional incorrect event on button 11 may be hardware in nature.

stefanoborini, 2019-05-29:
I have a keybow that started to behave erratically with button 11. I am not an efficient lua programmer so initially I thought the problem was in the code. After a while I realised I could replicate the behavior by pressing the buttons harder or simply touching the keybow on the side, or slightly pulling the USB cable.

I did some investigation and, I might be wrong, but it seems that the black metallic pylon with the nut is short circuiting the contacts of key 11 right and key 8 left. The nut is not a zero ohm connection but it does conduct, especially when mechanically pressed.

Am I the only one experiencing these problems?

stefanoborini, 2019-05-29:
On the topic, I just noticed that the instructions talk about plastic nuts, but I have metal ones.

stefanoborini, 2019-05-29:
I fixed the problem by replacing the metal nut with a plastic ring from the core tube of a pen. 

[image]

dglaude, 2019-05-29:
I believe this place is for code related issues. Pimoroni forum or support might be more appropriate. Maybe this could be closed now?
Thanks for the tip anyway.

stefanoborini, 2019-05-29:
@dglaude It can be closed, and I agree with you. I opened it for anyone that may think they have a software problem with button 11 and start searching for clues.

closed 2019-05-29 by stefanoborini