pimoroni/keybow-firmware #32: Unicode Character Input

TechnoElf, 2019-07-15:
I wanted to use the Keybow to allow me to easily type symbols from other languages (ᚠᚢᚦᚨᚱᚲ). Turns out, this is more difficult than expected as they are not part of the key codes set of a USB keyboard. These symbols are part of the Unicode character set but there is currently no way of typing these in the Keybow firmware. The USB HID documentation mentions a Unicode Page which could possibly be used to implement this.

Unicode input also has many other possible applications, such as math symbols, graphical symbols for ASCII art and emojis (Just imagine being able to type 💩 with the press of a single key!).

Gadgetoid, 2019-07-23:
I think Unicode characters can be entered using a key combination? I'm not sure if you could even successfully use these characters in the lua files, but it seems worth a try.

Gadgetoid, 2019-07-23:
Sorry- replied to this before I'd properly looked over your PR.

I believe the "Unicode Page" that you mention may be from Host->Device for displaying unicode characters on a screen or otherwise. As far as I can determine there is no way - short of a proprietary extension to HID with custom drivers - to facilitate Unicode character input.

There's some good (if outdated) talk about it here, including a message from someone actually involved in defining the HID spec suggesting using "vendor defined" usages - https://github.com/kiibohd/KiiConf/issues/30

closed 2023-01-20 by Gadgetoid