pimoroni/keybow-firmware #34: generic MIDI message sending

flauta, 2019-07-28:
I'm interested in extending the MIDI function, enabling the Keybow to send all kind of MIDI messages besides the NoteOn/NoteOff.
I'm not a C programmer at all, so my attached example code could be totally unadequate: take it only as an example of what I'd like to see it working.

`void sendMIDImessage(int channel, int code, int b1, int b2, int b3) {
	unsigned char buf[4];
	buf[0] = code | channel & 0x7f;
	buf[1] = b1 & 0x7f;
	buf[2] = b2 & 0x7f;
	buf[3] = b3 & 0x7f;
    write(midi_output, buf, 4);
}

void sendMIDINote(int channel, int note, int velocity, int state) {
    unsigned char code
    if(state==1){
		code = 0x90;
	}
	else
	{
		code = 0x80;
	}
	sendMIDImessage(channel, code, note, velocity, 0);
}`

Gadgetoid, 2019-08-20:
I'm totally onboard with this- I'm not sure why I started specifically with note-on/off rather than just a generic MIDI message (probably because I wanted to keep my POC super simple) but there's no reason why this couldn't work.

Do you know if there's actually an upper limit on the number of data bytes you can send over MIDI, or do things like PC, CC, etc, just all happen to fit into 3? I'd wager some third party products have a pretty liberal interpretation of the spec these days, unless they just run the bus fast and stuff everything into SysEx?

flauta, 2019-08-21:
Yes, as I read in the original MIDI specs, the single MIDI message can be only 2, 3 or 4 bytes (comprising the first code specifying the message type) 
see [https://www.usb.org/sites/default/files/midi10.pdf] - page 16
In my example to make the message sending simpler and to send every time 4 bytes, I was thinking that when the message should be 2 or 3 bytes, the extra byte(s) sent would not corrupt the stream. But I was not able to cross-compile my code and to test it: maybe the length of the buffer must  really be adapted each time to the MIDI specs.

closed 2023-01-20 by Gadgetoid