pimoroni/keybow-firmware #35: Loading pattern broken?

jezdez, 2019-08-02:
Hey there! Maybe I'm missing something but trying the following on a Keybow mini didn't load the patterns as expected:

```lua
keybow.load_pattern("rainbow.png")
```

```lua
keybow.load_pattern("patterns/rainbow.png")
```

```lua
keybow.load_pattern("patterns/rainbow")
```

```lua
local pattern = require "patterns/rainbow.png"
keybow.load_pattern(pattern)
```

Could someone explain what the correct syntax is for this please?

Gadgetoid, 2019-08-20:
Near as I can tell, there's nothing wrong with what you're attempting and a cursory check over the code doesn't yield any obvious problems. I'll try and run a debug build and see if I can replicate.

jezdez, 2019-08-20:
But to clarify, it's supposed to be `keybow.load_pattern("patterns/rainbow.png")` right?

Gadgetoid, 2019-08-20:
Ooof, lack of documentation hurts here- it should be:

```
keybow.load_pattern("patterns/rainbow")
```

The `png` part is, for better or worse, added automatically.

FleaRex, 2019-08-21:
Hi both!

This is my minimal working version:

```
require "keybow"

function handle_key_00(pressed)
    if pressed then
        load_new_pattern()
    end
end

function handle_key_01(pressed)
    if pressed then
        load_new_pattern()
    end
end

function handle_key_02(pressed)
    if pressed then
        load_new_pattern()
    end
end

function handle_key_03(pressed)
    if pressed then
        load_new_pattern()
    end
end

function handle_key_04(pressed)
    if pressed then
        load_new_pattern()
    end
end

function handle_key_05(pressed)
    if pressed then
        load_new_pattern()
    end
end

function handle_key_06(pressed)
    if pressed then
        load_new_pattern()
    end
end

function handle_key_07(pressed)
    if pressed then
        load_new_pattern()
    end
end

function handle_key_08(pressed)
    if pressed then
        load_new_pattern()
    end
end

function handle_key_09(pressed)
    if pressed then
        load_new_pattern()
    end
end

function handle_key_10(pressed)
    if pressed then
        load_new_pattern()
    end
end

function handle_key_11(pressed)
    if pressed then
        load_new_pattern()
    end
end

function load_new_pattern()
    keybow.load_pattern('patterns/rainbow')
end
```
Two quick Qs
1. Would you like a pull request to add this as an example?
2. It seems like [`patterns/tropical.png`](https://github.com/pimoroni/keybow-firmware/blob/v0.0.4/sdcard/patterns/tropical.png) isn't right in the most recent release. Is it worth doing a new release?

Gadgetoid, 2019-08-27:
Clearly we need some comprehensive documentation on this functionality- 

`keybow.auto_lights(false)` completely disables the image-based animation sequencing and light patterns in favour of setting individual LEDs manually.

FleaRex, 2019-08-28:
Thanks for confirming! Is the C code that the lua code calls around anywhere? It would be interesting to see what `keybow.load_pattern` and `keybow.auto_lights` really do. Either way I'm loving it and now that I know what I'm doing it's working really well.

Gadgetoid, 2019-11-05:
Absolutely- all the C source is here to browse: https://github.com/pimoroni/keybow-firmware/tree/master/keybow

Here's what `keybow.load_pattern` boils down to:

https://github.com/pimoroni/keybow-firmware/blob/7f359e567860eff871da96bbbad98a3eaac0377c/keybow/lua-config.c#L358-L374

Which, in turn, calls `read_png_file` in C:

https://github.com/pimoroni/keybow-firmware/blob/fd7b19a8818c11fd300d37ba20e83db542aac2ec/keybow/lights.c#L19-L84

And `keybow.auto_lights`:

https://github.com/pimoroni/keybow-firmware/blob/7f359e567860eff871da96bbbad98a3eaac0377c/keybow/lua-config.c#L301-L307