pimoroni/keybow-firmware #36: Keybow needs to be rebooted after detach/attach

andrewcarson, 2019-08-29:
I am using the Keybow attached to a powered USB hub.

When I detach and then reattach the hub from the computer (Windows 10), the Keybow does not re-register. I need to detach and reattach the Keybow itself (i.e. reboot it).

This is with the pre-built image from the [sdcard folder](https://github.com/pimoroni/keybow-firmware/tree/81a0777724484d1993c43153746cbc5b02672c09/sdcard)

If I install Raspian (Buster) on the Keybow, then clone, build and run the keybow binary directly, the detach/reattach via the hub works as expected.

This suggests (to me) that there is something missing from the cut-down Linux.

arthurlt, 2019-08-30:
I believe I'm having a similar issue.

Gadgetoid, 2019-09-02:
This works fine if I power Keybow directly, so it must be some interaction between Keybow -> Hub -> Host. Possibly blipping the power supply enough to crash the Pi, but not reboot?

I'll go on the hunt for a powered hub (there must be a loose PiHub somewhere) and see if I can replicate.

Gadgetoid, 2019-09-02:
Tested with an ORICO HF9US and it works fine if I disconnect/reconnect the hub from the host PC.

I'm afraid this is more likely to be an unavoidable electrical issue- IE: disconnecting/reconnecting your hubs, despite them being powered, causes a blip on the voltage supply that's enough to destabalise but not reboot the Pi. It might be worth connecting an HDMI cable & monitor (if possible) and seeing if you get any system output during the disconnect/reconnect.

Another potential test would be to connect an external power supply and repeat the experiment- connect a good 5v microUSB supply to your keybow, then disconnect/reconnect your hub.

Most peripherals don't have an embedded Linux system running on them, so we're probably susceptible to power instabilities that don't affect keyboards/mice or simpler things.

RapidOwl, 2020-10-29:
This also occurs for me on both a powered USB socket and a USB port on the back of my monitor. I thought I'd solved it by moving it to the monitor slot and it was fine for a couple of days, but now it's started needing a reboot every time I power up my computer and because the USB cable is in the back of the monitor, I need to yank the micro-USB end, which seems fragile.

eXsoR65, 2020-10-29:
This would be expected as it loses data to the computer, those needing to power off and on the PI. What I did to avoid needing to plug and unplug the USB cable is to by a Powered USB HUB with independent power switch to each port.  (https://www.amazon.com/gp/product/B083XTKV8V/)

RapidOwl, 2020-10-29:
Surely that's the same as having it in a powered USB socket on my laptop? That solution didn't work for me.

eXsoR65, 2020-11-03:
No because it will not lose power which is what you want. When the computer is restarted you **want** to also power cycle the pi. 
The USB Hub I got I simply just hit the power bottom on and off.

Gadgetoid, 2020-11-03:
I'm going to try bumping the kernel version & software for the keybow firmware at some point.

If that fails to get the thing more tolerant of disconnects, perhaps it's worth exposing a "reboot" feature where holding down a button for long enough (assuming the whole software stack hasn't crashed and burned) will force reboot the system or reset the USB or something.

I think this is a solvable problem, but I'm stretched so thin right now you could tie string around my ankles and fly me.

eXsoR65, 2020-11-03:
Hey fair enough, If there anything I can do to help let me know. testing etc.

Gadgetoid, 2020-11-03:
I'm not holding out  hope that it will make any difference, but updating the kernel/system and adding in WiFi connectivity will at least make it possible to debug (in cases where a Zero W is used anyway) and find some kind of solution.

My biggest issue at the moment (aside from lack of time... it's 11:30 pm here right now!) is being unable to replicate the keybow USB hub problem. It probably doesn't help that I don't have or use a regular USB hub!

While I'm futzing about with stuff, you *could* try - https://gist.github.com/Gadgetoid/f2ff7f3854a7a102074e8daf458063f5

I don't recall if this build actually works at all - it's just taken me 2+ hours to replicate it since I had lost my updated kernel config.

Note: if this is an electrical issue with an unstable voltage or drop in voltage when disconnecting a USB hub.. then that could mean the Pi has frozen/kernel paniced and is not possible to recover without a power cycle. Difficult to tell without a connected display, or debug over WiFi.

@ everyone - do your light patterns - if you have any continuously running - still work when the keybow doesn't?

eXsoR65, 2020-11-04:
@Gadgetoid Kool I'm going to give this a try. what the user account and password?

eXsoR65, 2020-11-04:
okay so I was unable to repro the issue. I was monitoring the dmesg via ssh. 

```
[Nov 4 18:56] dwc2 20980000.usb: new device is high-speed
[  +0.035554] dwc2 20980000.usb: new address 12
[  +0.021971] configfs-gadget gadget: high-speed config #1: config
[  +0.006186] gs_console_connect: port num [0] is not support console
```

eXsoR65, 2020-11-04:
okay so the default lay works fine but my customer layout does not which is the only difference. not sure if i'm doing somehting wrong?? 

```
require "keybow"
require "snippets/win_snippets"

-- Custom Colors --

function setup()
    keybow.auto_lights(false)
    keybow.clear_lights()
    keybow.set_pixel(0, 200, 200, 200)
    keybow.set_pixel(1, 0, 255, 0)
    keybow.set_pixel(2, 190, 0, 252) 
    keybow.set_pixel(3, 124, 252, 0)
    keybow.set_pixel(4, 124, 252, 0)
    keybow.set_pixel(5, 255, 0, 0)
    keybow.set_pixel(6, 255, 90, 0)
    keybow.set_pixel(7, 255, 90, 0)
    keybow.set_pixel(8, 255, 90, 0)
    keybow.set_pixel(9, 255, 255, 255)
    keybow.set_pixel(10, 255, 255, 255)
    keybow.set_pixel(11, 255, 255, 255)
end

-- Variables --

desktop = [["C:\Program Files (x86)\Citrix\ICA Client\SelfServicePlugin\SelfService.exe" -qlaunch "Desktop"]]
search = [[site:*duckduckgo.com]]
terminal = [[wt]]

-- Custom Key Mapping --

function handle_key_00(pressed)
    if pressed then
        win_snippets.task_manager()
        keybow.set_pixel(0, 255, 255, 255)
    else
        keybow.set_pixel(0, 200, 200, 200)
    end
end

function handle_key_01(pressed)
    if pressed then
        keybow.set_modifier(keybow.LEFT_META, keybow.KEY_DOWN)
        keybow.tap_key("r", pressed)
        keybow.set_modifier(keybow.LEFT_META, keybow.KEY_UP)
        keybow.sleep(400)
        keybow.text(terminal)
        keybow.sleep(250)
        keybow.tap_enter()
        keybow.set_pixel(2, 255, 255, 255)
    else
        keybow.set_pixel(2, 0, 255, 0)
    end
end

function handle_key_02(pressed)
    if pressed then
        keybow.set_modifier(keybow.LEFT_META, keybow.KEY_DOWN)
        keybow.tap_key("r", pressed)
        keybow.set_modifier(keybow.LEFT_META, keybow.KEY_UP)
        keybow.sleep(400)
        keybow.text(desktop)
        keybow.sleep(250)
        keybow.tap_enter()
        keybow.set_pixel(2, 255, 255, 255)
    else
        keybow.set_pixel(2, 190, 0, 252)
    end
end

function handle_key_03(pressed)
    if pressed then
        win_snippets.chrome_search(term)
        keybow.set_pixel(3, 255, 255, 255)
    else
        keybow.set_pixel(3, 124, 252, 0)
    end
end

function handle_key_04(pressed)
    if pressed then
        keybow.text(search)
        keybow.set_pixel(4, 255, 255, 255)
    else
        keybow.set_pixel(4, 124, 252, 0)
    end
end

function handle_key_05(pressed)
    if pressed then
        win_snippets.win_1()
        keybow.set_pixel(5, 255, 255, 255)
    else
        keybow.set_pixel(5, 255, 0, 0)
    end
end

function handle_key_06(pressed)
    if pressed then
        keybow.tap_key("4", pressed)
        keybow.set_pixel(6, 255, 255, 255)
    else
        keybow.set_pixel(6, 255, 255, 255)
    end
end

function handle_key_07(pressed)
    if pressed then
        keybow.tap_key("5", pressed)
        keybow.set_pixel(7, 255, 255, 255)
    else
        keybow.set_pixel(7, 255, 255, 255)
    end
end

function handle_key_08(pressed)
    if pressed then
        keybow.tap_key("6", pressed)
        keybow.set_pixel(8, 255, 255, 255)
    else
        keybow.set_pixel(8, 255, 255, 255)
    end
end

function handle_key_09(pressed)
    if pressed then
        keybow.tap_key("7", pressed)
        keybow.set_pixel(9, 255, 255, 255)
    else
        keybow.set_pixel(9, 255, 255, 255)
    end
end

function handle_key_10(pressed)
    if pressed then
        keybow.tap_key("8", pressed)
        keybow.set_pixel(10, 255, 255, 255)
    else
        keybow.set_pixel(10, 255, 255, 255)
    end
end

function handle_key_11(pressed)
    if pressed then
        keybow.tap_key("9", pressed)
        keybow.set_pixel(11, 255, 255, 255)
    else
        keybow.set_pixel(11, 255, 255, 255)
    end
end
```

eXsoR65, 2020-11-04:
Okay, so after some testing. The issue is with v0.0.4 only. With the build you provided I can't repro the issue.

Gadgetoid, 2020-11-04:
> Okay, so after some testing. The issue is with v0.0.4 only. With the build you provided I can't repro the issue.

That is *very* interesting! Looks like I need to check in my kernel configs, do a release with all the updates (and bonus WiFi for people brave enough to use it) and cross my fingers!

My intent with WiFi support was to bring luasockets into Keybow so you could send data from the keybow to a server, or vice versa. I didn't do all that much testing, though, and it remains to see how robust it will be. I'll probably shelve Luasockets until the updated kernel/initrd has bedded in and been tested for a while. One step at a time.

Good news if the new firmware both works and doesn't hang/crash. Thank you *very* much for testing this!

eXsoR65, 2020-11-04:
👍🏼 No problem, I'm no expert but just comparing. Notice that you have included new Device Tree Blob/overlays compared to v0.0.4.  In the build you provided it has bcm2708 and bcm2835 Device Tree Blob/overlays for both rpi-zreo and rpi-zero-w. ??