pimoroni/keybow-firmware #39: Document PNG to LED mapping. See #38

Gadgetoid, 2019-11-12:
The blog post linked in https://github.com/pimoroni/keybow-firmware/pull/38 makes a good point that our documentation lacks any explanation for how the pixels in PNG animations map to the LEDs on the keybow. We should fix this.

b099l3, 2021-02-05:
Hey I took a shot at trying to improve the documentation for the PNG to LED mapping in this [blog post](https://iainsmith.me/keybow-lights/)

Is there a place for the documentation?

I created these images to help understand the mapping
[image]
[image]

Extra-Fox, 2021-02-14:
Undocumented features are such a drag and while I really appreciate folks willing to take the time and energy to explore this, I'd be even more grateful for complete documentation from the OEM. I don't think it's enough to assume the "community" will help fill in the blanks.

closed 2023-01-20 by Gadgetoid