pimoroni/keybow-firmware #41: The keybow gets stuck after computer start

riskaone, 2020-03-05:
I have keybow mini. I use it with Window 10 and connect via OTG. When the computer starts, keybow starts as well. After the start-up period, the keybow have fixed colours on keys - not changing. It doesn't work when pressing keys. 

When I unplug and immediately plugin keybow, it works fine. I also realised that if I don't use keybow for a couple of hours, I have to again unplug it and plugin back to start working.

Gadgetoid, 2020-03-09:
Do you have a miniHDMI to HDMI adaptor and a display you can use to observe that the Pi Zero is outputting over HDMI when this happens? It's super hard to diagnose otherwise.

My gut feeling is that it's a power issue since USB ports were never intended to power embedded computers, but I don't have much to base that on.

Professor-X-Tech, 2020-08-18:
@riskaone I have this happen to me, it end up being a bad SD card. I would try another SD card if possible.

closed 2023-01-20 by Gadgetoid