pimoroni/keybow-firmware #42: Document build process from compilation -> initramfs

GothAck, 2020-03-24:
There are some build notes, and makefiles, but I don't see any documentation on how you compile this project into the initramfs in the sdcard folder. Maybe I'm missing something, but it would be useful to know how your process works end-to-end to be able to hack on this better.

SpikeTheMaster, 2020-06-14:
I would be interested in this also. My end goal is to get the keybow connected to my home network and to send MQTT messages to Home Assistant but I haven't yet worked out how to get the keybow to connect to WiFi.

Gadgetoid, 2020-06-15:
It's by no means documentation, but the toolchain used to build the Ramdisk is available here: https://github.com/pimoroni/rpi-ramdisk/tree/keybow  ... more or less.

It really needs retooling so that it can run on CI, and it looks like it's missing the file `keybow.service`.

It's been about a year since I touched it- I put the guts of Keybow into `/boot` so that the ramdisk doesn't need to change.

Gadgetoid, 2021-06-16:
Some slightly better instructions here, with a newer build process and the latest... "tweaks"... https://github.com/pimoroni/keybow-firmware/issues/77

Stuff is still a little broken and I don't have lots of time to fiddle about with it, but it more or less works.

closed 2023-01-20 by Gadgetoid