pimoroni/keybow-firmware #44: Shift and other modifier keys not working

dfirestorm, 2020-03-26:
Tried to assign the left side of the keyboard near WASD including left shift and left ctrl, did not work. Was not registered by reading all inputs either Issue was fixed by changing the keybow lua file to assign those keys to their HID usage keycodes (0xE0 to 0xE7). I also commented out the section later which uses those keys, not sure if that will cause issues.

eXsoR65, 2020-10-29:
I think it be best to share you share you keymap so that anyone may be able to help with this.

GwizBilly, 2021-01-15:
I got it working (Minecraft on Windows10 key mapping).
But you have to use set_modifier instead of set_key.
function handle_key_06(pressed)
    keybow.set_modifier(keybow.LEFT_SHIFT, pressed)
end

function handle_key_07(pressed)
    keybow.set_modifier(keybow.CTRL, pressed)
end

Here's a gist:
https://gist.github.com/GwizBilly/999bef86ba4a694111208bbda88de6fa

closed 2023-01-20 by Gadgetoid