pimoroni/keybow-firmware #55: Booting a keybow in USB device mode (ie: no sd card)

axelsimon, 2020-06-22:
Hi,
KeybowOS is designed to run entirely in RAM in order to reduce SD card wear down.
With this in mind, it would make sense to be able to use the Raspberry Pi Zero's [USB device boot mode](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/device.md), in order to skip the SD card entirely:
> When this boot mode is activated (usually after a failure to boot from the SD card), the Raspberry Pi puts its USB port into device mode and awaits a USB reset from the host. Example code showing how the host needs to talk to the Pi can be found [here](https://github.com/raspberrypi/usbboot).

ie : plug Keybow to computer with no SD card, Keybow goes into device mode, computer sends KeybowOS/firmware, Keybow boots and appears as USB device (keyboard).
One could even automate sending KeybowOS from the computer.

I had a look, and from what i can tell, there are some references to the SD card in the RAM FS used in KeybowOS:
the `init` in the `initrd` creates `/dev/mmcblk0p1` and then mounts it on `/boot`.
I'm pretty new at this, so is this a way to pretend there is an SD card with `mknod` and then using it? Or does this depend on the actual SD card being there to work?

Having more info on how `keybow-firmware` is built (cf. #42 ) would be helpful here.

Thanks!

Gadgetoid, 2020-06-23:
Keybow itself is actually run entirely out of /boot, the RAM disk image is *just* a cut down version of Raspbian with a startup script to run the keybow binary.

To be precise- the actual `keybow` executable is copied from `/boot` into the RAM disk by init before it's executed by the systemd unit. This is to allow easy software updates without having to rebuild and replace the entire initrd image.

init:

```
#!/bin/sh

sleep 0.2

mknod /dev/mmcblk0p1 b 179 1
mount -o rw /dev/mmcblk0p1 /boot

if [ -f "/boot/keybow" ]; then
        cp /boot/keybow /usr/bin/keybow
        chmod +x /usr/bin/keybow
        systemctl restart keybow
fi

exec /sbin/init
```

`etc/systemd/system/keybow.service`:

```
[Unit]
Description=Keybow Python Daemon
After=usb-gadget.service

[Service]
ExecStart=/usr/bin/keybow
Restart=on-failure
RestartSec=5

[Install]
WantedBy=default.target
```

The check for `/boot/keybow` in init might suggest that it fails over to a `usr/bin/keybow` inside the RAM disk image where the SD card is not available, but this (for better or worse) is not the case. Since Keybow relies upon the lua files and looks for them in `/boot` it'd probably fail even if it were in the RAM disk by default.

closed 2020-07-15 by Gadgetoid

axelsimon, 2020-07-15:
Hi @Gadgetoid, i really appreciate you answering and providing context.

Having now booted the Keybow from an SD card, i can infirm what i was saying previously: the Keybow never finishes booting when started in USB device mode and providing it a target directory from which to boot with `sudo ./rpiboot -d ../Keybow/keybow-0.0.4/`

Happy to try to tinker some more with this with a bit of guidance, but otherwise i don't think the original request has really been answered.