pimoroni/keybow-firmware #57: RPi image's initrd on https://github.com/pimoroni/keybow-firmware doesn't include the 'keybow' binary

BigValen, 2020-07-13:
I downloaded the files at https://github.com/pimoroni/keybow-firmware and wrote the 'sdcard' directory to an SD card. 

On booting the Pi Zero, the keybow service doesn't start. There is no /usr/bin/keybow file though the keybow.service systemd unit file is there, and trying to start /usr/bin/keybow. When I manually copied in the keybow binary, it complains that it can't find the keys.lua file.

I manually mounted /dev/mmc0blk0 under /boot, created a /boot/user directory and then copied the keys.lua file in there - then it worked. Unfortunately, manually hacking that into the initrd didn't work for me...somehow the EUID of root was 1001. This can only be built on a Debian system, yes ?

Gadgetoid, 2020-07-13:
`/usr/bin/keybow` is copied from `/boot` by `init`, it isn't baked into the `initrd` since that would make it a pain to ship updates (would require a complete `initrd` rebuild)

Not sure why you ran into this as an issue, since all the necessary files are in the sdcard directory. Were you attempting to build the `initrd` from scratch?

BigValen, 2020-07-13:
Ah. I had a look at the mount code - it assumes you put an MSDOS partition table on the sdcard, and a single partition. My SD Card was just a vfat filesystem, with no partition tables. I'm an idiot :)

Works flawlessly now.

closed 2020-07-13 by BigValen