pimoroni/keybow-firmware #58: Functions to trigger consecutive keystrokes not working

RapidOwl, 2020-08-07:
I've got a number of commands in my IDE that require consecutive keystrokes e.g. ctrl + T, T. I've tried the following, but only one keypress reaches my computer. Could anyone provide examples of how this could be done?

```
function handle_key_01(pressed)
    if pressed then
        keybow.set_modifier(keybow.LEFT_CTRL, keybow.KEY_DOWN)
        keybow.tap_key("T", pressed)
        keybow.sleep(500)
        keybow.tap_key("L", pressed)
        keybow.set_modifier(keybow.LEFT_CTRL, keybow.KEY_UP)
    end
end

function handle_key_02(pressed)
    if pressed then
        keybow.set_modifier(keybow.LEFT_CTRL, keybow.KEY_DOWN)
        keybow.tap_key("T", pressed)
        keybow.sleep(500)
        keybow.tap_key("T", pressed)
        keybow.set_modifier(keybow.LEFT_CTRL, keybow.KEY_UP)
    end
end
```

DanAtkinson, 2020-08-07:
I have something similar which might help.

```lua
function handle_key_00(pressed)
	-- Attach to worker process (Ctrl + Alt + P)
	if pressed then
		keybow.set_modifier(keybow.LEFT_CTRL, keybow.KEY_DOWN)
		keybow.sleep(100)
		keybow.set_modifier(keybow.LEFT_ALT, keybow.KEY_DOWN)
		keybow.sleep(100)
		keybow.tap_key("p")
		keybow.sleep(100)
		keybow.set_modifier(keybow.LEFT_ALT, keybow.KEY_UP)
		keybow.sleep(100)
		keybow.set_modifier(keybow.LEFT_CTRL, keybow.KEY_UP)
		keybow.sleep(500)
	end
end
```

RapidOwl, 2020-08-07:
I tried something similar to yours, but it didn't work:

Is there a way in Windows to show what keys have actually been "triggered"?

This was my code.

```
function handle_key_01(pressed)
    if pressed then
        keybow.set_modifier(keybow.LEFT_CTRL, keybow.KEY_DOWN)
        keybow.sleep(100)
        keybow.tap_key("T", pressed)
        keybow.sleep(100)
        keybow.tap_key("L", pressed)
        keybow.sleep(100)
        keybow.set_modifier(keybow.LEFT_CTRL, keybow.KEY_UP)
        keybow.sleep(500)
    end
end
```

DanAtkinson, 2020-08-07:
> Is there a way in Windows to show what keys have actually been "triggered"?

Type `osk` into Run (Win+R), or look for "On-Screen Keyboard".

RapidOwl, 2020-08-07:
Okay so I used my IDE's keystroke settings to try recording the keystrokes that the Keybow is sending and the above code is actually doing `ctrl + shift + T, L`.

Not sure why the Keybow is sending both `ctrl` and `shift` when the code is only triggering `ctrl`. I'm going to try adding a line to send `KEY_UP` for `shift` at the start of the function.

RapidOwl, 2020-08-07:
Putting `KEY_UP` for `shift` before the rest of the code did not help.

Both JetBrains Rider and Visual Studio receive `ctrl + shift + T, L` from the above code.

However the on screen keyboard only registers a long press from `ctrl`, but I could see `shift` flickering blue at one point.

I guess I could get around this by changing my keyboard shortcuts to not require consecutive keypresses, but I'm curious to know what's going on.

Professor-X-Tech, 2020-08-17:
@RapidOwl try creating a key combination function like below, I looked at the win_snippets.lua file for the idea. 

function keycombo()
	modifier(keybow.ESC, keybow.LEFT_CTRL, keybow.LEFT_SHIFT)
end

then assign is to a key 

function handle_key_01(pressed)
    if pressed then
        keybow.set_modifier(keycombo, keybow.KEY_DOWN)
        keybow.sleep(100)
        keybow.set_modifier(keycombo, keybow.KEY_UP)
    end
end

RapidOwl, 2020-08-18:
@Professor-X-Tech thanks, but all the `modifier` function does is wrap what I'm already doing.

My problem persists. The following code results in `shift` being pressed alongside all of the other keypresses.

```
function handle_key_01(pressed)
    if pressed then
        keybow.set_modifier(keybow.LEFT_CTRL, keybow.KEY_DOWN)
        keybow.tap_key("T", pressed)
        keybow.set_modifier(keybow.LEFT_CTRL, keybow.KEY_UP)
        keybow.sleep(100)
        keybow.tap_key("L", pressed)
    end
end
```

Gadgetoid, 2020-08-18:
`keybow.tap_key` doesn't have a second variable, I wonder if this is somehow causing your issue?

```lua
function keybow.tap_key(key)
    keybow.set_key(key, true)
    keybow.set_key(key, false)
end
```

I don't know quite how the Lua stack behaves, but it's possible this unconsumed additional argument is bleeding onto the stack and being popped into another function... it's a stretch!

RapidOwl, 2020-08-18:
@Gadgetoid I'm afraid that didn't help. There's got to be something really obvious and dumb that I'm doing, but I just can't see it.

Gadgetoid, 2020-08-18:
Or- just as likely- something less obvious and more dumb in my code 😆 

I wonder if I can test this in isolation.

RapidOwl, 2020-08-18:
If it helps, here's my `default.lua`. I've just been modifying this while experimenting. I've not removed any of the unused stuff in here so you can see it exactly as it is on my machine.

```
require "keybow"

-- Standard number pad mapping --

-- Key mappings --

-- keybow.set_key(keybow.ENTER, pressed)

function keycombo()
	modifier(keybow.ESC, keybow.LEFT_CTRL, keybow.LEFT_SHIFT)
end

function handle_key_00(pressed)
    if pressed then
        keybow.set_modifier(keybow.LEFT_SHIFT, keybow.KEY_DOWN)
        keybow.set_modifier(keybow.LEFT_CTRL, keybow.KEY_DOWN)
        keybow.set_modifier(keybow.LEFT_ALT, keybow.KEY_DOWN)
        keybow.tap_key("B")
        keybow.set_modifier(keybow.LEFT_SHIFT, keybow.KEY_UP)
        keybow.set_modifier(keybow.LEFT_CTRL, keybow.KEY_UP)
        keybow.set_modifier(keybow.LEFT_ALT, keybow.KEY_UP)
    end
end

function handle_key_01(pressed)
    if pressed then
        keybow.set_modifier(keybow.LEFT_CTRL, keybow.KEY_DOWN)
        keybow.tap_key("T")
        keybow.set_modifier(keybow.LEFT_CTRL, keybow.KEY_UP)
        keybow.sleep(100)
        keybow.tap_key("L")
    end
end

function handle_key_02(pressed)
    if pressed then
        keybow.set_modifier(keybow.LEFT_CTRL, keybow.KEY_DOWN)
        keybow.tap_key("T")
        keybow.set_modifier(keybow.LEFT_CTRL, keybow.KEY_UP)
        keybow.sleep(100)
        keybow.tap_key("T")
    end
end

function handle_key_03(pressed)
    keybow.set_key("1", pressed)
end

function handle_key_04(pressed)
    keybow.set_key("2", pressed)
end

function handle_key_05(pressed)
    keybow.set_key("3", pressed)
end

function handle_key_06(pressed)
    keybow.set_key("4", pressed)
end

function handle_key_07(pressed)
    keybow.set_key("5", pressed)
end

function handle_key_08(pressed)
    keybow.set_key("6", pressed)
end

function handle_key_09(pressed)
    keybow.set_key("7", pressed)
end

function handle_key_10(pressed)
    keybow.set_key("8", pressed)
end

function handle_key_11(pressed)
    keybow.set_key("9", pressed)
end
```

Gadgetoid, 2020-08-18:
Managed to shore myself up a better version of the local lua test.

You'll never guess what the problem is.

I'll give you a clue. The bug is so tiny and such a brutal footgun you can find it in this line:

```lua
keybow.tap_key("T")
```

RapidOwl, 2020-08-18:
Not sure I'm seeing it. Is it because I used a capital T so it assumed that shift is pressed?

Gadgetoid, 2020-08-18:
Yes! I'm facepalming pretty hard right now 😆 

It's such a gotcha,  because nobody would ordinarily expect that functionality to exist alongside the modifier stuff. Ugh.

RapidOwl, 2020-08-18:
Haha! I just looked through the other config files and noticed all the lower case. That'll teach me to take guesses at programming languages and platforms I don't know. Thanks for pointing it out. That'll save me setting up overloads in the IDE.

Gadgetoid, 2020-08-18:
No worries! I mean.. I wrote the thing and I still needed to rig tests and run the Lua before I finally realised what was going on 😆

closed 2023-01-20 by Gadgetoid