pimoroni/keybow-firmware #67: Can we use CAPSLOCK as a modifier?

kbullockuk, 2020-11-01:
I have tried to use the following to create a special key combination which is just CAPSLOCK & F12 (for use with Autohotkey) but the keybow doesn't appear to be sending CAPSLOCK, any chance you can make CAPSLOCK a modifier if it isn't already?

keybow.set_modifier(keybow.CAPSLOCK, keybow.KEY_DOWN)
keybow.tap_key(keybow.F12)
keybow.set_modifier(keybow.CAPSLOCK, keybow.KEY_DOWN)

Gadgetoid, 2020-11-01:
To preserve 12-key-rollover Keybow's `set_modifier` uses a bitmask set of modifier keys that's limited to 8, with the modifier keys indexed into it like so:

```
keybow.LEFT_CTRL = 0
keybow.LEFT_SHIFT = 1
keybow.LEFT_ALT = 2
keybow.LEFT_META = 3

keybow.RIGHT_CTRL = 4
keybow.RIGHT_SHIFT = 5
keybow.RIGHT_ALT = 6
keybow.RIGHT_META = 7
```

To press `CAPSLOCK` you have to treat it like an ordinary key:

```lua
keybow.set_key(keybow.CAPSLOCK, keybow.KEY_DOWN)
keybow.tap_key(keybow.F12)
keybow.set_key(keybow.CAPSLOCK, keybow.KEY_UP)
```

kbullockuk, 2020-11-01:
Superb thank you for your quick reply that has saved me a load of messing around!