pimoroni/keybow-firmware #77: documentation request: exact commands/build log for building the firmware

jakobi, 2021-06-14:
Documentation  request. 

I think the exact instructions to reducing the buster/raspbian  10 image for your 2020nov keybow firmware would be helpful. Maybe as simple  as the transscript/history from the image creation. 

AFAIS the github repo describing a generic pimoroni process for distro reduction seems to be configurable, and to start, I'd prefered something similar to your wifi firmware with as little
changes as possible, just being able to use nfs/ext2/ext4 and an editor command would 
have been plenty.

When creating my workaround-firmware-overlay tree (see other issue to be created in a minutes), I was struggling to repack the initrd. Having the transcript and the exact command would have saved me an hour.

(... | cpio -o -H newc | xz -9fc --check=crc32 --lzma2=dict=512KiB did rebuild the initrd and finally worked with the kernel from the above mentioned firmware image)

Thx
Peter

jakobi, 2021-06-14:
PS: can you release the beta wifi firmware in this repo, currently it's hidden in a gist, and I saw no mention of it anywhere...

Gadgetoid, 2021-06-16:
My build box used to test these steps is running Ubuntu 20.04. YMMV with something else.

You will probably need to follow the instructions in the README to import the relevant apt keys: https://github.com/gadgetoid/rpi-ramdisk/tree/keybow

Ignore the instructions about cloning all submodules, you'll only need the two mentioned below.

Assuming you have installed pydo with:

```bash
git clone git://github.com/ali1234/pydo
cd pydo && pip3 install .
```

And installed the dependencies:

```bash
sudo apt install libc6:i386 libstdc++6:i386 libgcc1:i386 \
                 libncurses5:i386 libtinfo5:i386 zlib1g:i386 \
                 build-essential git bc python zip wget gettext \
                 autoconf automake libtool pkg-config autopoint \
                 bison flex libglib2.0-dev gobject-introspection \
                 multistrap fakeroot fakechroot proot cpio \
                 qemu-user binfmt-support makedev \
                 gtk-doc-tools valac libssl-dev
```

You can build Keybow with something like:

```bash
git clone https://github.com/gadgetoid/rpi-ramdisk -b keybow
cd rpi-ramdisk
git submodule update --init --depth=1 kernel/linux
git submodule update --init --depth=1 firmware/usbboot
cp configs/keybow.config.py config.py
pydo --init
pydo :build -s10
```

I haven't done this for a *long* time, though, so we're behind a whole slew of kernel versions.

The WiFi firmware wasn't really in any state to widely release- since there's no huge demand for it I don't want to introduce the potential for breakage into "upstream" ... I wont have the time to deal with fallout.

This is the sort of thing I need to automate with a GitHub workflow, but I'd fear it will take forever and blow through a lot of usage minutes for not very much output.

Gadgetoid, 2021-06-16:
Note that you only want "keybow.img", which should replace "kernel.img" and "initrd" from the resulting build output, you will find these in the "boot" directory. There is still a lot of manual tinkering involved in the build process and no automation to get the full, final keybow firmware in one shot.

closed 2023-01-20 by Gadgetoid