pimoroni/keybow-firmware #78: Example code offering editor/hooks for persistance (working; but also feature request)

jakobi, 2021-06-14:
This is a feature request to add an editor to the 2020nov firmware image, and some hooks to permit persistance of modified files. 

An example implementing is attached to this issue.

Thanx,
Peter

References:

- the 2020nov firmware I'm building against: https://gist.github.com/Gadgetoid/f2ff7f3854a7a102074e8daf458063f5#file-keybow-wifi-zip

- overlay trees for above firmware and its initrd with initrd rebuild instructions
[MOD.txt](https://github.com/pimoroni/keybow-firmware/files/6651170/MOD.txt)
[MOD.zip](https://github.com/pimoroni/keybow-firmware/files/6651171/MOD.zip)

jakobi, 2021-06-14:
Post scriptum: sw/hw feature requests, severity "nice-to-have":

1) enabling the rpi zero to offer some directory as usb-gadget/usb-mass-storage :). I actually saw two nerves-based projects tearing off the rpi zero with a pico just to kind of "easier gain this capability". At least one with a nice pcb to map the pico into the pi's gpio pinout :).

2) I had to use HDMI for debugging boot up: a suitable HDMI cutout in the keybow acrylics  would have been nice, similar to micro-USB. W/o this cutout I was forced to unscrew the acrylics to connect the monitor to watch bootup of the keybow.

Gadgetoid, 2021-06-16:
>     1. enabling the rpi zero to offer some directory as usb-gadget/usb-mass-storage :). I actually saw two nerves-based projects tearing off the rpi zero with a pico just to kind of "easier gain this capability". At least one with a nice pcb to map the pico into the pi's gpio pinout :).

I spent countless hours trying to get Mass Storage to work in PiratePython. I wanted it to mount on *both* the users computer *and* the Linux system... which is not something filesystems are designed to do. It was (and still is) dicey and prone to errors.

Maybe it's possible to mount an alternate initrd and run an alternate config on boot that mounts /boot as Mass Storage when a key is held down, but otherwise just mounts internally.

We could mount a regular file as virtual storage, but it's still very tricky to sync changes and - iirc - only possible from host->device since there's no protocol to inform the host about changes to a mounted filesystem.

I think I see where you're going with your feature request/proof of concept and it aligns with something I was thinking today while trying to wrap my head around the Keybow firmware for probably the tenth time. It would be *nice* if the underlying system was pretty generic, and all the key components were modular. Of course I'm almost describing Tiny Core Linux at this point, but I always felt that went a little far in the other direction.

So hooks, I think, are a nice balance between the two. A stable, complete and working system that - if you want - offers the hooks to enable customization.

The `init` might just as well look for a file called "startup.sh" or "hook.sh" and then everything other than mounting/unmounting boot and exec'ing /sbin/init could be moved to there, making even the WiFi/SSH setup customizable.

I think I can tweak the rpi-ramdisk build process to set the correct permissions on `/etc/rc.local`, too.

Gadgetoid, 2021-06-16:
https://github.com/Gadgetoid/rpi-ramdisk/commit/8faee71e3264694d1d12cd1c7800d349620d9771

The TLDR is that this creates `/boot/startup` and any `.sh` files placed in that directory will be run by the preinit script.

Judging by the `cmdline.txt` it looks like it was intended for `/dev/mmcblk0p1` to be mounted as USB mass storage, but this does not appear to be working:

```
[   22.121387] Mass Storage Function, version: 2009/09/11
[   22.121449] LUN: removable file: (no medium)
[   22.121729] lun0: unable to open backing file: /dev/mmcblk0p1
[   22.122449] printk: console [ttyGS-1] disabled
[   22.122656] g_multi 20980000.usb: failed to start g_multi: -2
```

Presumably because it doesn't exist until created by `mknod /dev/mmcblk0p1 b 179 1`

jakobi, 2021-06-16:
### Wrt changing files on the keybow: 

- sshfs does work with the wifi firmware. Even editing a file on the host, then ssh root@keybowip pkill keybow should work.

For my own purpose, this should be enough to permit transient and persistant changes, in combination with xbindkeys/xdotool. 

Windows  should offer the same using e.g. autokeys with FUSE/SSHFS by way of  WSL  or https://github.com/billziss-gh/sshfs-win. 

### Mass storage on the keybow

I've had a short look at mass storage, and failed with g_mass_storage, but using the lower tier module libcomposite with configuration via sysfs did work to export a file as an usb storage device.

See scriptlet, adapted from https://unix.stackexchange.com/questions/388304/no-udc-shows-up-for-usb-mass-storage-gadget-with-configfs

[CONTAINERFS.script.zip](CONTAINERFS.script.zip)

However to trigger a keybow daemon restart could be a tad slow, with d0 maybe 30sec, and a linux PC waiting upto say 2 minutes for actual sync. 

A not quite-bash sketch could be this one (maybe this is similar to your piratepython mention?):

while true do
   : check the backing file timestamp
   d=time - stat (formtime) CONTAINERFS; 
   d>2*d0  || { restarting=0; continue; }
   d<d0 || { restarting=0; continue; }
   restarting!=1 || continue;
   restarting=1; 
   mount -oro,loop CONTAINERFS /mnt1; 
   cp /mnt1/lua-of-interest to lua-in-use-dir; 
   umount -fl /mnt1; 
   : do we need to expressly drop memory/vmcase -ve??
done

closed 2023-01-20 by Gadgetoid