pimoroni/keybow-firmware #80: Scroll Lock and Num Lock support

theboyknowsclass, 2021-09-19:
Hi, does anyone know of any way to emulate these keys?

I'm trying to add some extra controls to a KVM and those are quite often used

Gadgetoid, 2021-09-27:
Looks like they're just scancodes like everything else- https://gist.github.com/MightyPork/6da26e382a7ad91b5496ee55fdc73db2

Caps Lock is already implemented like so: https://github.com/pimoroni/keybow-firmware/blob/ec9bc9989935b196806e54065ca3ea8c67c738a6/sdcard/keybow.lua#L20

So maybe:
```lua
keybow.NUMLOCK = 0x43
keybow.SCROLLLOCK = 0x47
```

will work when coupled with the regular key press/release example here:

https://github.com/pimoroni/keybow-firmware/issues/67

```lua
keybow.set_key(keybow.NUMLOCK, keybow.KEY_DOWN)
keybow.tap_key(...)
keybow.set_key(keybow.NUMLOCK, keybow.KEY_UP)
```

theboyknowsclass, 2021-10-01:
hi @Gadgetoid - Thanks so much for this, it looks like I was using the wrong values - are you sure on these values?  scroll lock seems to work... but the value you posted for numlock is the same as F10?

weirdly. once I set these and have the following layout - the keybow seems to crash as soon as I hit the scrolllock (it triggers correctly the first time, but after that no keypresses work) - any ideas?

`require "keybow"

-- Standard number pad mapping --

-- Key mappings --

function handle_key_00(pressed)
    keybow.set_key(keybow.SCROLLLOCK, pressed)
end

function handle_key_01(pressed)
    keybow.set_key(keybow.NUMLOCK, pressed)
end`

theboyknowsclass, 2021-12-06:
anyone else tried this?  or can reproduce?

closed 2023-01-20 by Gadgetoid