pimoroni/keybow-firmware #81: Keyboard-Layout

Kuschizzza, 2022-01-11:
Hello!

I would like to use a german keyboard layout for the keybow. But unfortunately I don't have any clue how to do it.

I've seen the user "drop0x55" already made an awesome job with some languages:
[https://github.com/pimoroni/keybow-firmware/pull/63](url)

Maybe someone can help me how to start create a german keyboard layout. 
[http://kbdlayout.info/kbdgr](url)

Thanks a lot.

Kuschizzza, 2022-01-12:
I've already tried to make a german keyboad layout file, I had the norwegian keyboard as a template but after booting up the keybow, no inputs was recognize. I'm sure I made a couple of mistakes. 

`german = {}`

`KEYCODES             = {'a', 'b', 'c', 'd', 'e', 'f', 'g', 'h', 'i', 'j', 'k', 'l', 'm', 'n', 'o', 'p', 'q', 'r', 's', 't', 'u', 'v', 'w', 'x', 'y', 'z', '1', '2', '3', '4', '5', '6', '7', '8', '9', '0', '\n', '\a', '\b', '\t', ' ', '+', '\\', 'ü', '\a', '<', "#", 'ö', 'ä', '|', ',', '.', '-'}`
`SHIFTED_KEYCODES     = {'A', 'B', 'C', 'D', 'E', 'F', 'G', 'H', 'I', 'J', 'K', 'L', 'M', 'N', 'O', 'P', 'Q', 'R', 'S', 'T', 'U', 'V', 'W', 'X', 'Y', 'Z', '!', '"', '§', '$', '%', '&', '/', '(', ')', '=', '\a', '\a', '\a', '\a', '\a', '?', '\a', 'Ü', '\a' ,'>', '*', 'Ö', 'Ä', '°', ';', ':', '_'}`
`ALTGRD_KEYCODES      = {'•', '•', '•', '•', '€', '•', '•', '•', '•', '•', '•', '•', 'µ', '•', '•', '•', '@', '•', '•', '•', '•', '•', '•', '•', '•', '•', '•', '²', '³', '•', '•', '•', '{', '[', ']', '}', '•', '•', '•', '•', '•', '•', '•', '•', '•', '•', '•', '•', '•', '•', '•', '•', '•'}`
`SHIFTALTGRD_KEYCODES = {'•', '•', '•', '•', '€', '•', '•', '•', '•', '•', '•', '•', 'µ', '•', '•', '•', '@', '•', '•', '•', '•', '•', '•', '•', '•', '•', '•', '²', '³', '•', '•', '•', '{', '[', ']', '}', '•', '•', '•', '•', '•', '•', '•', '•', '•', '•', '•', '•', '•', '•', '•', '•', '•'}`

Kuschizzza, 2023-08-07:
-

Gadgetoid, 2023-08-07:
...

closed 2025-01-08 by Gadgetoid